vexide/vexide #32: Doesn't compile without critical_section feature

doinkythederp, 2024-04-13:
## Bug Description
Disabling the critical_section feature makes the crate not compile.

## Code to reproduce
N/A

## Expected vs. actual behavior
There shouldn't be any broken features.

## Additional information
- I have tested this issue on the latest development release.
- Rust version (see `rustc --version`): rustc 1.78.0-nightly (256b6fb19 2024-02-06)
- vexide version (see `Cargo.toml`): 273c7bffa9945d6d89c1321c3ae19a71279fe33d

Tropix126, 2024-04-13:
I think we should just remove the feature and just make this mandatory, since it's a critical part (pun intended) of our mutex implementation.

Gavin-Niederman, 2024-04-13:
> I think we should just remove the feature and just make this mandatory, since it's a critical part (queue laugh track) of our mutex implementation.

I agree; this was only ever a feature because it wasn't used internally.

PR #34 (2024-04-14): fix: remove critical-section feature
## Describe the changes this PR makes. Why should it be merged?
This PR completely removes the `critical-section` feature from vexide-core because the critical-section implementation is necessary for vexide-core internally.
This PR fixes #32.
## Additional Context
- These are breaking changes (semver: major).
closed 2024-04-14 by Gavin-Niederman