vexide/vexide #48: `AdiGyro` and `AdiServo` Device Support

Tropix126, 2024-04-17:
## What's the motivation for this feature?
Both of these devices are ones i've been avoiding for a while for various reasons, but we need to support them at some point.

## Describe the solution you'd like
- Add an `AdiGyro` support for abstracting over the old 3-wire yaw-rate gyro.
- Add an `AdiServo` struct for controlling the legacy cortex-era servos.

## Describe the drawbacks, if any
Both of these devices are very poorly documented in the SDK and may utilize private API functionality on VEXCode's side. Not sure.
- In `AdiGyro`'s case, VEXCode has an `is_calibrating` function while PROS does not. There's no provided function in the SDK for this functionality, meaning it's probably some undocumented status bit. Fun.
- PROS also is forced into immediately calibrating the sensor as soon as `vexDeviceAdiPortConfigSet` is called.
- In `AdiServo`'s case, PROS doesn't even support it at all. The units/range of the ADI setter is comlpetely unknown.

I also don't have access to any of this hardware to test on.

zabackary, 2024-11-16:
Can this be split up into two issues? I have physical access to a ADI yaw-rate gyroscope, but not the servo, so I can pick up the implementation for just `AdiGyro` sometime next week.

zabackary, 2024-11-17:
Here's PROS's implementation of the gyroscope for reference: https://github.com/purduesigbots/pros/blob/8b33c707445bd911a065793f5b867e2173359c61/src/devices/vdml_ext_adi.c#L362-L422

Also, here's the servo implementation: https://github.com/purduesigbots/pros/blob/8b33c707445bd911a065793f5b867e2173359c61/src/devices/vdml_ext_adi.c#L252-L274

`AdiServo` seems like it would be pretty easy to implement

Tropix126, 2024-11-17:
> Also, here's the servo implementation: [purduesigbots/pros@`8b33c70`/src/devices/vdml_ext_adi.c#L252-L274](https://github.com/purduesigbots/pros/blob/8b33c707445bd911a065793f5b867e2173359c61/src/devices/vdml_ext_adi.c#L252-L274)

This is the MC29 control code, which is analgous to vexide's `AdiMotor` struct. [This doc](https://www.robotc.net/files/pdf/vex-natural-language/hp_servo.pdf) makes me think that they might operate on 8-bit PWM (same as MC29s), but it's hard to tell.

closed 2024-11-29 by Tropix126