vexide/vexide #69: Add way to set text background color.

doinkythederp, 2024-05-25:
## What's the motivation for this feature?
It would be nice to set a background color when drawing Text to the screen. This way I can make my text a little more stylish, maybe improve contrast with dark foregrounds, etc.

## Describe the solution you'd like
One solution would be to add another method to Text that also takes `background` parameter (e.g. `Text::write`). Then, the Fill implementation would delegate out to that and choose a default background color.

## Describe the drawbacks, if any
This would mean you have to use a different method than other widgets to drawing Texts with backgrounds.

## Describe the alternative solutions, if any
Another option would be to add another parameter to the Fill trait.

## Additional context
The implementation should use `vexDisplayBackgroundColor` to set a background color before drawing text.

zabackary, 2024-11-22:
I'm going to go ahead and implement this with the solution you proposed, even though it's probably true that there's a more elegant solution. Maybe there should be another trait `FillText` which implements a `fill_text` method specifically for `Text` widgets?

zabackary, 2024-11-23:
Should there be a `fill_text` method on `Display` which takes `Text`, a foreground color, and background color?

doinkythederp, 2024-11-23:
I think that would be the best option, and would advocate for calling it `Display::write` which is a little more concise.

zabackary, 2024-11-23:
@Gavin-Niederman [mentioned](https://github.com/vexide/vexide/pull/247#pullrequestreview-2454937834) as a review comment that `write` might be a little ambiguous since it's completely unrelated to `vexide_core::io::Write::write`, so what do you think?

zabackary, 2024-11-23:
Also, should `Text` even implement `Fill` to begin with?

PR #247 (2024-11-29): feat: add a way to set `Text` background color
## Describe the changes this PR makes. Why should it be merged?

This PR adds the `Text::write` API for writing text to a `Display` with a background color.

This PR contains breaking changes. It changes the default background color of text written with the `Fill` trait from the last used background color (which is unpredictable and probably not the best default) to transparent.

## Additional Context

- I have tested these changes on a VEX V5 Brain.
- semver: major (see earlier). If you don't count that as a breaking change, then it's a minor cha [... cut]
commit 28d9faf2 (2024-11-29): feat: add a way to set `Text` background color (#247)

## Describe the changes this PR makes. Why should it be merged?

This PR adds the `Text::write` API for writing text to a `Display` with
a background color.

This PR contains breaking changes. It changes the default background
color of text written with the `Fill` trait from the last used
background color (which is unpredictable and probably not the best
default) to transparent.

## Additional Context

- I have tested these changes on a VEX V5 Brain.
- semver: major (see earlier). If you don't count that as a breaking
change, then it's a m [... cut]
closed 2024-11-29 by Tropix126