vexide/vexide #72: Add API for drawing aligned text

doinkythederp, 2024-05-25:
## What's the motivation for this feature?
Right now there's no way to draw right/bottom-aligned or horizontally/vertically centered text. Once text metrics are added, this could tediously be done without library support, but I figure it would be nicer to have a real API.

## Describe the solution you'd like
`Text` should store 2 new Copyable enums, named `VAlign` and `HAlign`, with the variants `Top`, `Center`, and `Bottom` as well as `Left`, `Center`, and `Right`. These would default to `Left` and `Top`, but could be changed via builder-style methods on `Text`. When `Text`s are drawn, they would take these into account to decide where to put the text relative to the origin point passed in the constructor.

Example:
```rs
let txt = Text::new("Hello world", TextSize::Medium, (50, 50))
    .align(VAlign::Center, HAlign::Right);
txt.fill(&mut screen, Rgb::BLUE);
```

[image]

## Describe the drawbacks, if any

I'm not sure if it's possible to calculate metrics for large or small text sizes.

## Describe the alternative solutions, if any
There should be a method on `Text` called `write_aligned` that's similar to `Fill::fill` but also has support for adding vertical and horizontal text alignments. This would involve adding 2 new Copyable enums, named `VAlign` and `HAlign`, with the variants `Top`, `Center`, and `Bottom` as well as `Left`, `Center`, and `Right`. These would be passed as arguments.

Example:
```rs
let txt = Text::new("Hello world", TextSize::Medium, (50, 50));
txt.write_aligned(&mut screen, Rgb::BLUE, VAlign::Center, HAlign::Right);
```

[image]

## Additional context

This proposal will likely require text metrics to be implemented first.

doinkythederp, 2024-05-27:
Blocked on #70 right now

alexDickhans, 2024-06-04:
I think I could do this one

closed 2024-06-08 by doinkythederp