vexide/vexide #92: Graphical panic messages do not appear in DoubleBuffered render mode

doinkythederp, 2024-06-09:
## Bug Description

After using `Screen::set_render_mode` to set the render mode to `DoubleBuffer`, graphical panic messages will not appear and it will appear as if the program is frozen upon a panic.

## Code to reproduce

```rs
#![no_std]
#![no_main]

use vexide::prelude::*;

#[vexide::main]
async fn main(mut p: Peripherals) {
    p.screen.set_render_mode(vexide::devices::screen::RenderMode::DoubleBuffered);
    panic!("something failed");
}
```

## Expected vs. actual behavior

The panic message should appear on screen in a red box, but instead nothing appears at all.

## Additional information
- vexide version (see `Cargo.toml`): `259590b0e06c60b0662d5e871a25abbea4ff4984`
- I have tested this issue on the latest development release.
- I would be willing to implement a fix for this issue.
- Priority this issue should have (low/med/high): med

alexDickhans, 2024-06-10:
From what I can tell this also happens with `Text` widgets

doinkythederp, 2024-06-10:
> From what I can tell this also happens with `Text` widgets

What you're describing might actually be the intended behavior outside of a panic: the double buffered render mode switches the screen into a state where all drawing is delayed until `Screen::render` is called. When a panic occurs, the handler doesn't call `Screen::render` so the message is never displayed.

PR #134 (2024-09-26): fix: stop double buffered rending from stopping panic message rendering
## Describe the changes this PR makes. Why should it be merged?

This PR fixes a bug where enabling screen double buffering would stop graphical panic messages from being displayed on the screen.

Closes #92 

## Additional Context
- I have tested these changes on a VEX V5 brain.
commit 9ca7cd5d (2024-09-26): fix: stop double buffered rending from stopping panic message rendering (#134)

## Describe the changes this PR makes. Why should it be merged?

This PR fixes a bug where enabling screen double buffering would stop
graphical panic messages from being displayed on the screen.

Closes #92 

## Additional Context
- I have tested these changes on a VEX V5 brain.
closed 2024-09-26 by doinkythederp