vexide/vexide #129: Slint platform does not work inside of async runtime

doinkythederp, 2024-09-07:
## Bug Description

The Slint platform does not work inside a future being run by block_on (such as `main`). It stalls while trying to run block_on inside of block_on.

## Code to reproduce
```rs
use vexide::prelude::*;

slint::include_modules!();

#[vexide::main]
async fn main(p: Peripherals) {
    initialize_slint_platform(p.screen);
    let ui = AppWindow::new().unwrap();
    ui.run().unwrap();
}
```

## Additional information
- I have tested this issue on the latest development release.
- Rust version (see `rustc --version`): rustc 1.78.0-nightly (256b6fb19 2024-02-06)
- Priority this issue should have (low/med/high): high as it's likely a bug with the async executor

Gavin-Niederman, 2024-10-06:
I cannot reproduce this on main.
This is the testing code I used, it's slightly different because I didn't want to mess with slint build scripts for fonts but it achieves the same result:
```rust
slint::slint! {
    import { Button, VerticalBox } from "std-widgets.slint";
    export component AppWindow inherits Window {
        background: #000;
        Button {
            clicked => { }
        }
    }
}

#[vexide::main]
async fn main(p: Peripherals) {
    initialize_slint_platform(p.screen);
    let ui = AppWindow::new().unwrap();
    ui.run().unwrap();
}
```

I can even run tasks while slint is running:

```rust
#[vexide::main]
async fn main(p: Peripherals) {
    spawn(async {
        loop {
            sleep(Duration::from_millis(250)).await;
            println!("Running while slint is running");
        }
    }).detach();
    initialize_slint_platform(p.screen);
    let ui = AppWindow::new().unwrap();
    ui.run().unwrap();
}
```

Can you please try this again and see if it is actually not working?

doinkythederp, 2024-10-06:
Likely fixed by #144

closed 2024-10-06 by doinkythederp