vexide/vexide #131: Controller API sucks to use.

Tropix126, 2024-09-13:
## Problem Description
Really not sure how else to put it - most of this is my fault. On paper it *seems* fine, but there is a lot of redundant error handling involved with the controller API as well as some other minor annoyances:
- Controller errors for a given timeslice cannot occur for one channel but not another, yet each getter returns its own `Result`, as if the operation could fail for individual buttons.
- Joystick data is returned as `f32`, which doesn't match `Motor::set_voltage`, making a cast necessary to spin motors (the most common operation done with controller data).
- We need some sort of event loop system for detecting press/release.

## Solutions

I think it might be worth considering returning all controller state as a single struct like `pros-rs` used to do:

```rs
let state = controller.state()?;

for motor in self.left_motors.iter_mut() {
    motor.set_voltage(state.left_stick.y * Motor::MAX_VOLTAGE)?;
}
for motor in self.right_motors.iter_mut() {
    motor.set_voltage(state.right_stick.y * Motor::MAX_VOLTAGE)?;
}
```

Although i'm still debating on whether or not this is a good idea. I think other rust libraries for gamepad/controller support have a similar scheme, so it's really tempting. No other device really does this though.

Aztro-dev, 2024-09-15:
Why does `Motor::set_voltage` even require double precision anyway? There's no way that a motor can respond to nanovolt differences.

Tropix126, 2024-09-25:
> Why does `Motor::set_voltage` even require double precision anyway? There's no way that a motor can respond to nanovolt differences.

The SDK internally expects a `double` with `vexDeviceMotorVoltageSet`. That and general API consistency (most floating-point device methods in the SDK are double-precision).

Gavin-Niederman, 2024-10-08:
> No other device really does this though.
Distance also does this. Regardless, I think returning all controller state at one time is a better API.

PR #152 (2024-10-10): refactor: switch to an all in one controller state getter
## Describe the changes this PR makes. Why should it be merged?
This PR refactors the controller API so that all controller state is returned at once. A single controller channel cannot error without every other controller channel erroring, so this removes all unnecessary error handling.

Closes #131 

## Additional Context
- I have tested these changes on a VEX V5 brain.
- These are breaking changes (semver: major).
commit 08be3c02 (2024-10-10): refactor: switch to an all in one controller state getter (#152)

## Describe the changes this PR makes. Why should it be merged?
This PR refactors the controller API so that all controller state is
returned at once. A single controller channel cannot error without every
other controller channel erroring, so this removes all unnecessary error
handling.

Closes #131 

## Additional Context
- I have tested these changes on a VEX V5 brain.
- These are breaking changes (semver: major).
closed 2024-10-10 by Tropix126