vexide/vexide #253: IMU calibration still times out sometimes.

Tropix126, 2024-12-07:
## Bug Description
There's some weird hardware stuff that I don't fully understand going on here. IMU calibration can timeout once, twice, or even three times in a row and not successfully occur. I think we need to look more into the cause of this or remove the timeouts all together since this isn't uncommon.

zabackary, 2025-01-10:
I learned recently that VEXos (or is it vexOS?)'s auto-calibration feature runs at program start (at least, the `is_calibrating` status bit is set for the first couple of seconds of program run so I assume that's what's happening), so it's possible that asking the inertial to calibrate during that time is running into some weird timing-related bugs on the inertial firmware side which is causing this.

Tropix126, 2025-01-26:
(probably) closed by #279

closed 2025-01-26 by Tropix126