vexide/vexide #270: Cannot build using rust-nightly

jacobjuneau6, 2025-01-03:
## Bug Description
The command ''' cargo v5 build ''' does not build. This was done on an ubuntu 24.10 system with the rust toolchain installed following the instructions on Jan 3rd, 2025. The code is vexide-template.

## Code to reproduce
se vexide-template

## Expected vs. actual behavior
The program should compile. It does not.

## Additional information
logs
'''(base) [jacob@archlinux vexide-template]$ cargo v5 build
info: syncing channel updates for 'nightly-x86_64-unknown-linux-gnu'
info: latest update on 2025-01-03, rust version 1.85.0-nightly (4363f9b6f 2025-01-02)
info: component 'rust-src' is up to date
info: downloading component 'llvm-tools'
info: installing component 'llvm-tools'
error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `/home/jacob/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/rustc - --crate-name ___ --print=file-names --target /home/jacob/vexide-template/armv7a-vex-v5.json --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=split-debuginfo --print=crate-name --print=cfg` (exit status: 1)
  --- stderr
  error: Error loading target specification: ARM targets must specify their float ABI. Run `rustc --print target-list` for a list of built-in targets

(base) [jacob@archlinux vexide-template]$'''
Move all applicable items out of the comment:
- Operating system (Linux/macOS/Windows): Linux
- vexide version (see `Cargo.toml`):
- Rust version (see `rustc --version`):
- Priority this issue should have (high):
'''
$ rustc --version
rustc 1.85.0-nightly (4363f9b6f 2025-01-02)
'''

Gavin-Niederman, 2025-01-03:
This is an issue with our target spec on newer versions of rust. Were working on it. Thanks for reporting! :D

jacobjuneau6, 2025-01-03:
Is there a version I can downgrade to temporarily to get compilation working?

Gavin-Niederman, 2025-01-03:
Yes! nightly-2024-12-06 should work.

jacobjuneau6, 2025-01-03:
That version works. Thank You.

Tropix126, 2025-01-27:
Fixed in `cargo-v5` 0.10.0.

closed 2025-01-27 by Tropix126