vexide/vexide #291: vexide template fails to build

jacobjuneau6, 2025-01-28:
## Bug Description
git clone the vexide template and compile it.
error
```error: the `#[global_allocator]` in vexide_core conflicts with global allocator in: vexide_core```

## Code to reproduce
the vexide template

## Expected vs. actual behavior
The program should compile. It gives this error 
```error: the `#[global_allocator]` in vexide_core conflicts with global allocator in: vexide_core```

## Additional information
- Operating system (Linux)
- vexide version (0.6.0)
- Rust version (```rustc 1.86.0-nightly (2f348cb7c 2025-01-27)```)
- Priority this issue should have (high)
- I have tested this issue on the latest development release.

Gavin-Niederman, 2025-01-29:
I will fix this upstream but for now can you try deleting your Cargo.lock and then rebuilding?

Tropix126, 2025-01-31:
Fixed by https://github.com/vexide/vexide-template/pull/12 (I believe)

closed 2025-01-31 by Tropix126