vexide/vexide #315: crt0 implementation in `vexide-startup`.

Tropix126, 2025-03-19:
## What's the motivation for this feature?

Some people may want to link in a `libc` (such as newlib) to their vexide program for compatibility with some larger C libraries. At the moment this won't work because global constructors are not called by `vexide-startup`. The current workaround for this is to manually invoke `__libc_init_array` within your program's `main` function before accessing any C stuff over FFI.

## Describe the solution you'd like
Add an **optional** `crt0` feature to `vexide-startup` that invokes global constructors/destructors through `__libc_init_array`/`__libc_fini_array`. This won't be a default feature.

Tropix126, 2025-03-29:
Closing for now as we'd need some extensive linkerscript changes. Might revisit later.

closed 2025-03-29 by Tropix126

lewisfm, 2025-10-14:
Here is an implementation of what we would need to do: https://github.com/vexide/VexSwift/blob/master/Sources/VexV5Support/Startup.c#L17

lewisfm, 2025-10-14:
Here are the ARM docs on what our linker script needs: <https://developer.arm.com/documentation/107976/20-1-0/Map-code-and-data-to-your-target-device/Linker-scripts/Linker-script-Section-names?lang=en>

Some highlights... we would need to change the linker script to support:

1. [Thread-local](https://gcc.gnu.org/onlinedocs/gcc/Thread-Local.html) sections (this is different from task locals), which are `.tbss` and `.tdata`. These are recognized by the compiler because they are marked with the [`SHF_TLS`](https://www.sco.com/developers/gabi/2003-12-17/ch4.sheader.html#sh_flags) flag.

   Since we are a single-threaded platform, the current thread-local block will never actually change. But, it's not uncommon for C globals (e.g. "errno") to be defined as thread-locals for compatibility reasons, so they will need to be placed in these sections.

   We will also need to tell the compiler about our thread-local model, which determines how thread-locals are looked up at runtime. There is no single good source of information about this, so here are some resources:

   picolibc docs which say they use "local-exec": <https://github.com/picolibc/picolibc/blob/main/doc/tls.md>
   [ARM CLI flag docs](https://developer.arm.com/documentation/107976/20-1-0/Clang-reference/clang-command-line-options/-ftls-model?lang=en) which recommend using "local-exec" for the most part.
   [ARM C startup docs](https://developer.arm.com/documentation/107976/20-1-0/C-and-C---libraries/picolibc-C-runtime?lang=en) which describe how TLS is laid out

	At least in the "local-exec" mode, the way TLS variables are actually resolved at runtime is by calling [this intrinsic](https://github.com/picolibc/picolibc/blob/02c895f9f29a0e349aacef2c92481ca0a6d163e4/newlib/libc/picolib/machine/arm/read_tp.S#L54) to get a pointer to get the TLS block and adding whatever offset the variable is from this.

2. (Optionally), the GOT, which is used to implement position-independent code. We don't need support for that, but it's enabled by default in compilers like clang, so it's not out of the question for a C library we link with to use it.

Here is the info on what crt0 startup means for embedded ARM (the ARM Embedded Toolchain uses picolibc): <https://developer.arm.com/documentation/107976/20-1-0/C-and-C---libraries/Picolibc-startup-code?lang=en>

lewisfm, 2025-10-14:
Reopening this because we might want to do it now that we have std support