vexide/vexide #332: Printing to the controller screen in a loop is potentially broken.

Tropix126, 2025-05-13:
## Bug Description
Thought that we dealt with this one in #154, but people are supposedly running into trouble when printing to the controller in loops still. I will need to test this further to confirm.

## Expected vs. actual behavior
See the screenshot below.

[image]

This could potentially indicate either a race condition (compiler optimizations speeding up the printing process), or UB at the worst case.

zabackary, 2025-05-13:
It's funny how I keep finding these ones. I guess the "safe" code I write is bad practice. 

By the way, the bug I fixed in the future implementation that I haven't had the time to try to merge into vexide is that the success check is inverted so it only polls once. Seems to happen a lot :)

zabackary, 2025-05-13:
Currently I just print in a loop with `try_set_text` which works fine.